jgerstmayr/EXUDYN #4: ModuleNotFoundError occurs: No module named 'exudynCPP'

hunter1992, 2021-07-03:
Hello jgerstmayr:
Thanks for your job! I got an import error before using the EXUDYN.

After installing EXUDYN using the method in section 1.2.2 of theDoc, 
ModuleNotFoundError occurs: No module named 'exudynCPP' .

My environment:
```
PS C:\...\EXUDYN-master\main\pythonDev> python
Python 3.8.8 (default, Apr 13 2021, 15:08:03) [MSC v.1916 64 bit (AMD64)] :: Anaconda, Inc. on win32

Warning:
This Python interpreter is in a conda environment, but the environment has
not been activated.  Libraries may fail to load.  To activate this environment
please see https://conda.io/activation

Type "help", "copyright", "credits" or "license" for more information.
```

Then, I tried to run the pytest.py file under the path:  EXUDYN-master\main\pythonDev
```
PS C:\Users\zhm\Documents\ZHM\Scripts\Python\EXUDYN-master\main\pythonDev> python .\pytest.py
Traceback (most recent call last):
  File "C:\Users\zhm\Documents\ZHM\Scripts\Python\EXUDYN-master\main\pythonDev\exudyn\__init__.py", line 16, in <module>
    from .exudynCPP import *
ModuleNotFoundError: No module named 'exudyn.exudynCPP'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File ".\pytest.py", line 14, in <module>
    import exudyn as exu
  File "C:\Users\zhm\Documents\ZHM\Scripts\Python\EXUDYN-master\main\pythonDev\exudyn\__init__.py", line 19, in <module>
    from exudynCPP import *
ModuleNotFoundError: No module named 'exudynCPP'
```

I suspected that I did not install the whl file correctly, so I installed it again, but it turned out that there was nothing wrong with the installation of the whl file:
```
PS C:\Users\zhm\Documents\ZHM\Scripts\Python\EXUDYN-master\main\dist> pip install .\exudyn-1.0.248-cp38-cp38-win_amd64.whl 
Processing c:\users\zhm\documents\zhm\scripts\python\exudyn-master\main\dist\exudyn-1.0.248-cp38-cp38-win_amd64.whl
exudyn is already installed with the same version as the provided wheel. Use --force-reinstall to force an installation of the wheel.
```

I was confused. 
How should it be solved?

hunter1992, 2021-07-04:
OK, the problem was solved. (Under Windows10)
The python packge "exudyn" needs a file named "exudynCPP.cp38-win_amd64.pyd" which is complied by Visual Studio.

The steps I did:
1. Make sure the Microsoft C++ build Tools has been installed correctly;
2. Under the path: **_EXUDYN/main/_**, where has setup.py file, run the command "python setup.py install" in PowerShell or Anaconda Prompt, the building process will start, just wait it down;
3. After the build process, there is a new subpath under **_EXUDYN/main/_** ;
4. The python package folder "exudyn" locate at **_EXUDYN/main/build/lib.win-amd64-3.8_** , and the .pyd file "**exudynCPP.cp38-win_amd64.pyd**" is there;
5. If you wanna run the "pytest.py" file under **_EXUDYN/main/pythonDev_** path, you need to copy the  exudynCPP.cp38-win_amd64.pyd file into the "exudyn" folder under **_EXUDYN/main/pythonDev_**.

I prefer that exudyn can be successfully deployed on Linux, because it is more convenient and direct to use python and C++ on Linux. 
But when I try to compile the exudynCPP.xyz.so file under Linux (Manjaro), I always encounter a segment error when I run the pytest.py file. 
I hope that friends who know about this can solve this problem.

That's all.

jgerstmayr, 2021-07-04:
Hi hunter1992! Thanks for your post and congratulations that it worked finally on Windows.
It seems that you did not correctly install the windows wheel (using pip). There are several options for installation on Windows as well as Linux given in theDoc.pdf, Sections 1.2.2, 1.2.3 (install), 1.2.5 (build under Windows), 1.2.7 (build under Linux). The errors you got look like if you have not installed but just copied the files into local directories. 
After the step 2 described in your previous post,  "python setup.py install", exudyn should be already installed and you should be able to run exudyn in Python directly. 
The library has been reportedly tried by dozens of users, without problems. The only known issue is with CPUs that are not capable of AVX-2. 
Regarding the Linux problems, please try to use the steps described in Section 1.2.7. Also check docs/howTo files, especially "ubuntuPythonSetup.txt", which may contain valuable information. I only test with Ubuntu18.04 and 20.04, but have no experience with Manjaro. If you can provide me more information on the segmentation fault, I could help you.

jgerstmayr, 2021-07-08:
Corrected a couple of issues with Linux compilation (which may have the reason for the above mentioned problems), therefore this issue should be resolved. Tests were successful with Python 3.6 and 3.8 on Ubuntu.

closed 2021-07-08 by jgerstmayr

jgerstmayr, 2021-07-12:
UPDATE: there were still some issues with GLFW in the recent versions. Version 1.0.276 now tested with UBUNTU20.04, running smoothly for several examples, including graphics interface (but needs numpy, matplotlib, tkinter, etc. installed).