jgerstmayr/EXUDYN #5: some examples solving  failed

oridong, 2021-07-25:
Microsoft Windows [Version 10.0.19043.1110]
Python 3.9.5 (tags/v3.9.5:0a7dcbd, May  3 2021, 17:27:52) [MSC v.1928 64 bit (AMD64)] on win32

Compiling by `python setup.py build` 

many similar prints  like the following
`C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\1 /c /nologo /Ox /W3 /GL /DNDEBUG /MD -DVERSION_INFO="1.0.276" -Id:\vsc_dyn\ex3.9.egg\pybind11\include -Isrc -Iinclude -Iinclude/pybind11 -Iinclude/lest -I
-IE:\ZAProgramsInstalled\Python39\include -IE:\ZAProgramsInstalled\Python39\iicrosoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\ATLMFC\includoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\include -IC:\Progrclude\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0s (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windwinrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt Data.cpp /Fobuild\temp.win-amd64-3.9\Release\src/Objects/CNodeGenericData.obj /GL /W3 /Gy /Zc:wchar_t /Zi /Gm- /O2 /sdl /Zc:inline /fp:precise /D _MBCS /DGS /errorReport:prompt /WX- /Zc:forScope /Gd /Oy /Oi /openmp /std:c++17 /FC /E /arch:AVX2
CNodeGenericData.cpp`

then add `D:/vsc_dyn/EXUDYN/main/build/lib.win-amd64-3.9/` to .pth file in python directory site-packages.

expamples which  success , are   `main\pythonDev\Examples\3SpringsDistance.py`  `ANCF_slidingJoint2D.py`  `ANCF_slidingJoint2Drigid.py` `cartesianSpringDamper.py`  `addPrismaticJoint.py` `addRevoluteJoint.py` `ANCF_cantilever_test_dyn.py` `ANCF_cantilever_test.py` `ANCF_contact_circle.py` `ANCF_contact_circle2.py`  `ANCF_slidingJoint2D.py` `ANCF_slidingJoint2Drigid.py` `ANCF_tests2.py` `ANCFALEtest.py`
`bicycleIftommBenchmark.py` `cartesianSpringDamper.py` `coordinateSpringDamper.py` `finiteSegmentMethod.py` `flexibleRotor3Dtest.py` `geneticOptimizationExample.py`  `geneticOptimizationSliderCrank.py` `interactiveTutorial.py` `lavalRotor2Dtest.py` `leggedRobot.py`  `multiMbsTest.py`  `multiprocessingTest.py` `myFirstExample.py`

`ALEANCF_pipe.py`    `ANCF_moving_rigidbody.py`  `ANCF_switchingSlidingJoint2D.py`

print the following 

> Exception has occurred: ValueError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
> SolveDynamic terminated
>   File "D:\vsc_dyn\EXUDYN\main\build\lib.win-amd64-3.9\exudyn\solver.py", line 249, in SolveDynamic
>     raise ValueError("SolveDynamic terminated")

ANCF_test_halfcircle.py  
> Exception has occurred: ValueError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
> SolveStatic terminated due to errors
>   File "D:\vsc_dyn\EXUDYN\main\build\lib.win-amd64-3.9\exudyn\solver.py", line 165, in SolveStatic

massSpringFrictionInteractive.py

> Exception has occurred: TypeError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
> UFspring() takes 9 positional arguments but 10 were given
>   File "D:\vsc_dyn\EXUDYN\main\build\lib.win-amd64-3.9\exudyn\interactive.py", line 425, in InitializeSolver
>     self.mbs.sys['solver'].InitializeSolver(self.mbs, self.simulationSettings)
>   File "D:\vsc_dyn\EXUDYN\main\build\lib.win-amd64-3.9\exudyn\interactive.py", line 281, in __init__
>     self.InitializeSolver() #solver gets ready to be called repeatedly

oridong, 2021-07-25:
after I changed python 3.9 to anaconda3 2020.02 python3.7.6  and re-compile ,all examples succeeded. after I changed python 3.9 to anaconda3 2020.02 python3.7.6  and re-compile ,all examples succeeded. Perhaps, ngsolve's interface to python is only valid for python 3.7 ,thus libngpy could be imported correctly . 

there are still failure examples:

> Exception has occurred: RuntimeError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
> Unable to cast Python instance to C++ type (compile in debug mode for details)
>   File "D:\vsc_dyn\EXUDYN\main\py37build\lib.win-amd64-3.7\exudyn\robotics.py", line 325, in CreateRedundantCoordinateMBS
>     visualization=VObjectJointGeneric(axesRadius = 0.01,axesLength=0.1, color=color4red)))
>   File "D:\vsc_dyn\EXUDYN\main\pythonDev\Examples\serialRobotOldTests.py", line 283, in <module>
>     robotDict = newRobot.CreateRedundantCoordinateMBS(mbs, loadJointUserFunctionslist, baseMarker)

> Exception has occurred: ValueError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
> SolveDynamic terminated
>   File "D:\vsc_dyn\EXUDYN\main\py37build\lib.win-amd64-3.7\exudyn\solver.py", line 249, in SolveDynamic
>     raise ValueError("SolveDynamic terminated")
>   File "D:\vsc_dyn\EXUDYN\main\pythonDev\Examples\SliderCrank.py", line 130, in <module>
>     exu.SolveDynamic(mbs, simulationSettings)

> Exception has occurred: ValueError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
> SolveDynamic terminated
>   File "D:\vsc_dyn\EXUDYN\main\py37build\lib.win-amd64-3.7\exudyn\solver.py", line 249, in SolveDynamic
>     raise ValueError("SolveDynamic terminated")
>   File "D:\vsc_dyn\EXUDYN\main\pythonDev\Examples\sliderCrankCMSacme.py", line 457, in <module>
>     exu.SolveDynamic(mbs, simulationSettings)

> Exception has occurred: ValueError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
> SolveDynamic terminated
>   File "D:\vsc_dyn\EXUDYN\main\py37build\lib.win-amd64-3.7\exudyn\solver.py", line 249, in SolveDynamic
>     raise ValueError("SolveDynamic terminated")
>   File "D:\vsc_dyn\EXUDYN\main\pythonDev\Examples\solutionViewerTest.py", line 138, in <module>
>     exu.SolveDynamic(mbs, simulationSettings, showHints=True)

jgerstmayr, 2021-07-27:
> after I changed python 3.9 to anaconda3 2020.02 python3.7.6 and re-compile ,all examples succeeded. after I changed python 3.9 to anaconda3 2020.02 python3.7.6 and re-compile ,all examples succeeded. Perhaps, ngsolve's interface to python is only valid for python 3.7 ,thus libngpy could be imported correctly .
> 
> there are still failure examples:
> 
> > Exception has occurred: RuntimeError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
> > Unable to cast Python instance to C++ type (compile in debug mode for details)
> > File "D:\vsc_dyn\EXUDYN\main\py37build\lib.win-amd64-3.7\exudyn\robotics.py", line 325, in CreateRedundantCoordinateMBS
> > visualization=VObjectJointGeneric(axesRadius = 0.01,axesLength=0.1, color=color4red)))
> > File "D:\vsc_dyn\EXUDYN\main\pythonDev\Examples\serialRobotOldTests.py", line 283, in 
> > robotDict = newRobot.CreateRedundantCoordinateMBS(mbs, loadJointUserFunctionslist, baseMarker)
> 
> > Exception has occurred: ValueError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
> > SolveDynamic terminated
> > File "D:\vsc_dyn\EXUDYN\main\py37build\lib.win-amd64-3.7\exudyn\solver.py", line 249, in SolveDynamic
> > raise ValueError("SolveDynamic terminated")
> > File "D:\vsc_dyn\EXUDYN\main\pythonDev\Examples\SliderCrank.py", line 130, in 
> > exu.SolveDynamic(mbs, simulationSettings)
> 
> > Exception has occurred: ValueError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
> > SolveDynamic terminated
> > File "D:\vsc_dyn\EXUDYN\main\py37build\lib.win-amd64-3.7\exudyn\solver.py", line 249, in SolveDynamic
> > raise ValueError("SolveDynamic terminated")
> > File "D:\vsc_dyn\EXUDYN\main\pythonDev\Examples\sliderCrankCMSacme.py", line 457, in 
> > exu.SolveDynamic(mbs, simulationSettings)
> 
> > Exception has occurred: ValueError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
> > SolveDynamic terminated
> > File "D:\vsc_dyn\EXUDYN\main\py37build\lib.win-amd64-3.7\exudyn\solver.py", line 249, in SolveDynamic
> > raise ValueError("SolveDynamic terminated")
> > File "D:\vsc_dyn\EXUDYN\main\pythonDev\Examples\solutionViewerTest.py", line 138, in 
> > exu.SolveDynamic(mbs, simulationSettings, showHints=True)

Thanks for using Exudyn! As mentioned in theDoc:
- VS2019 did not work so far due to problems with Eigen / Sparse. This would explain some examples not running
- Python 3.7 is recommended under Windows; Python 3.8 also has been tested
- Python 3.9 has been successfully tested under Linux but not under Windows. 
- NGsolve has only been tested under Python 3.7 under Windows and MacOS

> Microsoft Windows [Version 10.0.19043.1110]
> Python 3.9.5 (tags/v3.9.5:0a7dcbd, May 3 2021, 17:27:52) [MSC v.1928 64 bit (AMD64)] on win32
>
> Compiling by `python setup.py build`
>
> many similar prints [...]

Thanks for using Exudyn! As mentioned in theDoc:
- VS2019 did not work so far due to problems with Eigen / Sparse. This would explain some examples not running
- Python 3.7 is recommended under Windows; Python 3.8 also has been tested
- Python 3.9 has been successfully tested under Linux but not under Windows. 
- NGsolve has only been tested under Python 3.7 under Windows and MacOS

jgerstmayr, 2021-07-27:
> after I changed python 3.9 to anaconda3 2020.02 python3.7.6 and re-compile ,all examples succeeded. after I changed python 3.9 to anaconda3 2020.02 python3.7.6 and re-compile ,all examples [...]

The error in SolveDynamic always occurs, because this is the interface to C++ and problems occur somewhere inside C++. In order to understand the reasons for the exceptions, I would need more details of the error messages. Furthermore, it is very helpful to know how you installed the lib. Did you use "python setup.py install"? Why don't you just install the wheels for your Anaconda (as recommended in theDoc.pdf?)

Thanks a lot for helping to improve Exudyn!

oridong, 2021-07-29:
Thank you for EXUDYN, I have another question, how can I study these theroy in EXUDYN? what book could you recommend for me ? I have finished 'Ahmed A. Shabana - Dynamics of Multibody Systems-4th.pdf'  and I am reading `Dynamics of Flexible Multibody Systems Rigid Finite Element Method` , but I am still perplexed. There are several methods for `Dynamics of Multibody System` .

jgerstmayr, 2021-07-29:
Moved this question to Discussion/Theory !

closed 2021-11-23 by jgerstmayr