jgerstmayr/EXUDYN #8: Compile error

yumianhuli2, 2021-11-23:
Hello!
I installed all the dependent libraries by the web Install documentation,But got the following error:
OS:ubuntu18.04

XXX：~/XXXE/XUDYN/main$ sudo python3 setup.py install
platform == Linux
python version = 3.6.9
build Exudyn version 1.1.26
***************************
add flag PERFORM_UNIT_TESTS
***************************

running install
running bdist_egg
running egg_info
writing pythonDev/exudyn.egg-info/PKG-INFO
writing dependency_links to pythonDev/exudyn.egg-info/dependency_links.txt
writing top-level names to pythonDev/exudyn.egg-info/top_level.txt
reading manifest file 'pythonDev/exudyn.egg-info/SOURCES.txt'
writing manifest file 'pythonDev/exudyn.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.6m -c /tmp/tmpvitybp7p.cpp -o tmp/tmpvitybp7p.o -std=c++17
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.6m -c /tmp/tmp2kggs69t.cpp -o tmp/tmp2kggs69t.o -fvisibility=hidden
building 'exudyn.exudynCPP' extension
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION_INFO="1.1.26" -I/home/kakuhin/Downloads/EXUDYN/main/.eggs/pybind11-2.6.0-py3.6.egg/pybind11/include -Isrc -Iinclude -Iinclude/pybind11 -Iinclude/lest -Iinclude/glfw/deps -Iinclude/glfw -I/usr/include/python3.6m -c src/Autogenerated/versionCpp.cpp -o build/temp.linux-x86_64-3.6/src/Autogenerated/versionCpp.o -Wno-comment -Wall -DEXUDYN_RELEASE -Wno-non-template-friend -std=c++17 -fvisibility=hidden
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DVERSION_INFO="1.1.26" -I/home/kakuhin/Downloads/EXUDYN/main/.eggs/pybind11-2.6.0-py3.6.egg/pybind11/include -Isrc -Iinclude -Iinclude/pybind11 -Iinclude/lest -Iinclude/glfw/deps -Iinclude/glfw -I/usr/include/python3.6m -c src/Graphics/GlfwClient.cpp -o build/temp.linux-x86_64-3.6/src/Graphics/GlfwClient.o -Wno-comment -Wall -DEXUDYN_RELEASE -Wno-non-template-friend -std=c++17 -fvisibility=hidden
In file included from src/Linalg/BasicLinalg.h:31:0,
                 from src/Graphics/GraphicsData.h:21,
                 from src/Graphics/GlfwClient.h:28,
                 from src/Graphics/GlfwClient.cpp:14:
src/Linalg/ResizableMatrix.h: In member function ‘void ResizableMatrixBase<T>::Swap(ResizableMatrixBase<T>&)’:
src/Linalg/ResizableMatrix.h:171:13: error: ‘data’ was not declared in this scope
   std::swap(data, other.data);
             ^~~~
src/Linalg/ResizableMatrix.h:171:13: note: suggested alternative:
In file included from /usr/include/c++/7/string:51:0,
                 from /usr/include/c++/7/stdexcept:39,
                 from src/Utilities/ReleaseAssert.h:22,
                 from src/Graphics/GlfwClient.h:17,
                 from src/Graphics/GlfwClient.cpp:14:
/usr/include/c++/7/bits/range_access.h:318:5: note:   ‘std::data’
     data(initializer_list<_Tp> __il) noexcept
     ^~~~
In file included from src/Linalg/BasicLinalg.h:31:0,
                 from src/Graphics/GraphicsData.h:21,
                 from src/Graphics/GlfwClient.h:28,
                 from src/Graphics/GlfwClient.cpp:14:
src/Linalg/ResizableMatrix.h:172:13: error: ‘numberOfRows’ was not declared in this scope
   std::swap(numberOfRows, other.numberOfRows);
             ^~~~~~~~~~~~
src/Linalg/ResizableMatrix.h:173:13: error: ‘numberOfColumns’ was not declared in this scope
   std::swap(numberOfColumns, other.numberOfColumns);
             ^~~~~~~~~~~~~~~
In file included from src/Linalg/BasicLinalg.h:32:0,
                 from src/Graphics/GraphicsData.h:21,
                 from src/Graphics/GlfwClient.h:28,
                 from src/Graphics/GlfwClient.cpp:14:
src/Linalg/LinkedDataMatrix.h: In member function ‘void LinkedDataMatrixBase<T>::Swap(LinkedDataMatrixBase<T>&)’:
src/Linalg/LinkedDataMatrix.h:141:13: error: ‘data’ was not declared in this scope
   std::swap(data, other.data);
             ^~~~
src/Linalg/LinkedDataMatrix.h:141:13: note: suggested alternative:
In file included from /usr/include/c++/7/string:51:0,
                 from /usr/include/c++/7/stdexcept:39,
                 from src/Utilities/ReleaseAssert.h:22,
                 from src/Graphics/GlfwClient.h:17,
                 from src/Graphics/GlfwClient.cpp:14:
/usr/include/c++/7/bits/range_access.h:318:5: note:   ‘std::data’
     data(initializer_list<_Tp> __il) noexcept
     ^~~~
In file included from src/Linalg/BasicLinalg.h:32:0,
                 from src/Graphics/GraphicsData.h:21,
                 from src/Graphics/GlfwClient.h:28,
                 from src/Graphics/GlfwClient.cpp:14:
src/Linalg/LinkedDataMatrix.h:142:13: error: ‘numberOfRows’ was not declared in this scope
   std::swap(numberOfRows, other.numberOfRows);
             ^~~~~~~~~~~~
src/Linalg/LinkedDataMatrix.h:143:13: error: ‘numberOfColumns’ was not declared in this scope
   std::swap(numberOfColumns, other.numberOfColumns);
             ^~~~~~~~~~~~~~~
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

jgerstmayr, 2021-11-23:
Sorry for that issue, we are only running Windows tests per default, but will fix this issue soon.
In the meantime, you could use pre-compiled Linux Exudyn 1.1.2 version from the main/dist folder.

yumianhuli2, 2021-11-23:
> Sorry for that issue, we are only running Windows tests per default, but will fix this issue soon. In the meantime, you could use pre-compiled Linux Exudyn 1.1.2 version from the main/dist folder.

Ok, thanks! Looking forward to your update！

jgerstmayr, 2021-11-23:
Added some minor fixes (VisualStudio 2017 is more relaxed with some unconventional code than gcc). Still some warnings, but compilation and execution tested on Ubuntu18.04 and Ubuntu20.04. Should be resolved now.

closed 2021-11-23 by jgerstmayr