jgerstmayr/EXUDYN #11: Error while running resultsMonitor.py: module 'signal' has no attribute 'getsignal'

ManuelZ, 2022-02-08:
I'm getting the following error when running the `resultsMonitor.py` script:

```
Traceback (most recent call last):
  File "exudyn\resultsMonitor.py", line 257, in <module>
    plt.pause(updatePeriod)
  File "C:\Users\Manuel\anaconda3\lib\site-packages\matplotlib\pyplot.py", line 548, in pause
    canvas.start_event_loop(interval)
  File "C:\Users\Manuel\anaconda3\lib\site-packages\matplotlib\backends\backend_qt.py", line 408, in start_event_loop
    with _maybe_allow_interrupt(event_loop):
  File "C:\Users\Manuel\anaconda3\lib\contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "C:\Users\Manuel\anaconda3\lib\site-packages\matplotlib\backends\qt_compat.py", line 218, in _maybe_allow_interrupt
    old_sigint_handler = signal.getsignal(signal.SIGINT)
AttributeError: module 'signal' has no attribute 'getsignal'
```

To reproduce:

Change directory into `pythonDev`:
```
cd EXUDYN\main\pythonDev
```

Run the genetic optimizer:
```
python Examples\geneticOptimizationSliderCrank.py
```

In another tab run the `resultsMonitor.py` script:
```
python exudyn\resultsMonitor.py solution\geneticSliderCrank.txt
```

I think this may be due to the clash between the [built-in signal package](https://docs.python.org/3/library/signal.html#signal.getsignal) which does have the `getsignal` function and the Exudyn file called `signal.py`, which causes Matplotlib to find the Exudyn's signal package instead of the Python's one.

jgerstmayr, 2022-02-09:
Did you change Examples\geneticOptimizationSliderCrank.py ? on my installation it runs. I would need more information on Python version, etc. 
do you use "from exudyn import *" ? you should not use that. This may be the reason.
In fact, one could omit the signal module, as it is mostly not needed. If it makes problems, I will rename that in future. In my case, it is imported as exu.signal, which does not cause any problems.

ManuelZ, 2022-02-10:
I didn't modify `Examples\geneticOptimizationSliderCrank.py`.

The `geneticOptimizationSliderCrank.py` example  has these imports 
```
from exudyn.itemInterface import *
from exudyn.utilities import *
```

But no `from exudyn import *`.

My system:
```
>>> import platform; print(platform.platform())
Windows-10-10.0.19041-SP0

>>> import sys; print("Python", sys.version)
Python 3.8.8 (default, Apr 13 2021, 15:08:03) [MSC v.1916 64 bit (AMD64)]

>>> import exudyn
>>> exudyn.__version__
'1.1.42'

>>> import matplotlib
>>> matplotlib.__version__
'3.5.1'
```

jgerstmayr, 2022-02-10:
This is due to Python 3.8.8. 
The command
  python.exe exudyn\resultsMonitor.py Examples\solution\geneticSliderCrank.txt
from runs fine from command line with Python 3.8.5. 

However, I get the same error in Python 3.9.7. The resultsMonitor.py is not included in the TestSuite. However, if you have a older Python 3.6 or 3.7 installed in parallel to Python 3.8.8, you could install exudyn there and use this Python to view your results as is does not have to be the same Python version ...

**exudyn.signal will be removed in next version of exudyn**

NOTE: In your description, you were missing "Examples\" in your path.