jgerstmayr/EXUDYN #13: Question: How to detect broken mechanism

ManuelZ, 2022-02-10:
Hi, sometimes during an optimization procedure with a Genetic algorithm, the chosen set of parameters makes my mechanism break.

Normal mechanism:
[image]

Broken mechanism:
[image]

What could I do to either, detect this and stop the simulation, or to prevent some joints from going out of safe ranges?

jgerstmayr, 2022-02-10:
First, your solver may not converge or have large drift (in case of TrapezoidalInde2 solver). You can use a preStepUserFunction to perform checks (iterating over all relevant joints) at the beginning of every step to detect if some safe ranges are left. The preStepUserFunction, even if written in Python, is usually fast because it is only called once per step.

ManuelZ, 2022-02-10:
Ok thanks. I'm doing this inside my `SetPreStepUserFunction`
```
u = mbs.GetObjectOutput(joint_idx, exu.OutputVariableType.Rotation)
```

However, two things:
1. I'm using a `ObjectJointRevolute2D`, and in the documentation, the point 7.2.33.1 (Description) is empty (other joints do have this section filled, so that I can know what `OutputVariableType`s are available for this joint).
2. Trying `exu.OutputVariableType.Position` I get "ObjectJointRevolute2D has no OutputVariableType 'Position'". Trying `exu.OutputVariableType.Rotation` gets me `CObjectJointRevolute2D::GetOutputVariableConnector not implemented`.

Does this mean that isn't currently possible to get output varibles from a `ObjectJointRevolute2D` in this way? Or should I add a sensor first?

jgerstmayr, 2022-02-28:
The CObjectJointRevolute2D has currently no output variables. But the values can be easily computed from the associated bodies. The rotation is not available in JointRevolute2D, as the 2D revolute joint can also be attached to position-based markers, which have no rotation. The documentation for this joint is indeed missing, but you can see the implemented methods in the C++ code.

closed 2022-02-28 by jgerstmayr