jgerstmayr/EXUDYN #15: ModuleNotFoundError: No module named 'exudynCPP'

hunter1992, 2022-03-27:
Hello Gerstmayr!
Thanks for your work!

There is a problem in the process of installing EXUDYN. I need your help.

**Here is my installing steps:**

1. Decompressing the code package downloaded from github, I got the "EXUDYN-master" directory;

2. Under "EXUDYN-master/main" path, I run the command: python setup.py build, and everything is going well.

3. Still under "EXUDYN-master/main" path, I run the command: python setup.py install, no errors occured, 
    and the EXUDYN python lib has been installed under my home local path (I think):
[image]

**The problem occurred when I was using exudyn**:
I wanna try the exudyn lib by running the ANCF_cantilever_test.py from the "EXUDYN-master/main/pythonDev/Examples/" path,
the Error occurs: No module named 'exudynCPP'. 

This is the error message:

        Traceback (most recent call last):
          File "/home/zhm/Documents/Scripts/Python/EXUDYN-master/main/pythonDev/exudyn/__init__.py", line 16, in <module>
            from .exudynCPP import *
        ModuleNotFoundError: No module named 'exudyn.exudynCPP'
        
        During handling of the above exception, another exception occurred:
        
        Traceback (most recent call last):
          File "/home/zhm/Documents/Scripts/Python/EXUDYN-master/main/pythonDev/ANCF_cantilever_test.py", line 14, in <module>
            import exudyn as exu
          File "/home/zhm/Documents/Scripts/Python/EXUDYN-master/main/pythonDev/exudyn/__init__.py", line 19, in <module>
            from exudynCPP import *
        ModuleNotFoundError: No module named 'exudynCPP'

Please tell me how to get the 'exudynCPP' module.

hunter1992, 2022-03-27:
Same solution like under Windows10 which I posted before.

jgerstmayr, 2022-03-28:
I think that you do not have to copy the binary files (.pyd or similar) to the exudyn site-packages, if you run the command (as suggested in theDoc.pdf):
  python setup.py bdist_wheel 
and then run 
  pip install --user dist/exudyn[newly created wheel name]

This will install exudyn in your active python environment. It then includes the binary files as well. 
I do not see the purpose of the --prefix option: do you intend anything different from --user in pip?

hunter1992, 2022-03-29:
Yeah! It works :)  
Thank you so much!

Thank you for your contribution to helping more researchers use and learn the ANCF method!

closed 2022-03-29 by jgerstmayr

dummifiedme, 2022-08-24:
> I think that you do not have to copy the binary files (.pyd or similar) to the exudyn site-packages, if you run the command (as suggested in theDoc.pdf): python setup.py bdist_wheel and then run pip [...]

After posting #24 and #25, I tried this, and it failed with an error upon running the "bdist_wheel":
```
In file included from src/Main/rendererPythonInterface.h:24,
                 from src/Graphics/GlfwClient.h:19,
                 from src/Graphics/GlfwClient.cpp:14:
src/Utilities/ResizableArray.h: In constructor ‘ResizableArray<T>::ResizableArray(ResizableArray<T>&&)’:
src/Utilities/ResizableArray.h:162:27: error: ‘exchange’ is not a member of ‘std’
  162 |                 data(std::exchange(other.data, nullptr)),
      |                           ^~~~~~~~
src/Utilities/ResizableArray.h:163:39: error: ‘exchange’ is not a member of ‘std’
  163 |                 maxNumberOfItems(std::exchange(other.maxNumberOfItems, 0)),
      |                                       ^~~~~~~~
src/Utilities/ResizableArray.h:164:36: error: ‘exchange’ is not a member of ‘std’
  164 |                 numberOfItems(std::exchange(other.numberOfItems, 0)) {}
      |                                    ^~~~~~~~
src/Utilities/ResizableArray.h: In member function ‘ResizableArray<T>& ResizableArray<T>::operator=(ResizableArray<T>&&)’:
src/Utilities/ResizableArray.h:326:37: error: ‘exchange’ is not a member of ‘std’
  326 |                         data = std::exchange(other.data, nullptr);
      |                                     ^~~~~~~~
src/Utilities/ResizableArray.h:327:49: error: ‘exchange’ is not a member of ‘std’
  327 |                         maxNumberOfItems = std::exchange(other.maxNumberOfItems, 0);
      |                                                 ^~~~~~~~
src/Utilities/ResizableArray.h:328:46: error: ‘exchange’ is not a member of ‘std’
  328 |                         numberOfItems = std::exchange(other.numberOfItems, 0);
      |                                              ^~~~~~~~
error: command '/usr/bin/gcc' failed with exit code 1
```

**System Info:**
Manjaro Linux
Python 3.10.5