jgerstmayr/EXUDYN #18: An error in "geometricallyExactBeam3Dtest.py"

Jiang-Cui, 2022-06-21:
Dear professor,

I run "geometricallyExactBeam3Dtest.py", but an error occurs as follows:

>File "main\pythonDev\TestModels\geometricallyExactBeam3Dtest.py", line 55, in <module>
>    sectionData = exu.BeamSection()
>AttributeError: module 'exudyn' has no attribute 'BeamSection'

Could you help me solve the question? Thank you!

jgerstmayr, 2022-06-22:
First I assume, that you are not using the latest version of exudyn, which you can get with:
pip install exudyn --pre

However, as said at the very beginning of ObjectBeamGeometricallyExact3D in theDoc.pdf:
"OBJECT UNDER CONSTRUCTION"

It will take some weeks from now to be completed.

Jiang-Cui, 2022-06-22:
Thank you, Professor Gerstmayr! It can work!

closed 2022-06-22 by jgerstmayr