jgerstmayr/EXUDYN #24: Unable to install from pip

dummifiedme, 2022-08-24:
I am trying to install exudyn from pip but it keeps failing with an error. I was trying with `pip install exudyn`.

```
ERROR: Could not find a version that satisfies the requirement exudyn (from versions: none)
ERROR: No matching distribution found for exudyn
```
What should I do? Do I have to build it from source?

PS. I am on Manjaro Linux (ArchBased) and I prefer using python without Anaconda.

jgerstmayr, 2022-08-24:
Sorry, but it seems that your linux system is not supported by pre-installed wheels (currently Ubuntu with Python 3.6 and Python 3.8; but the compiled wheels usually also work on other platforms). If you share your Python version, I will think about building further Python/linux versions in the future.

dummifiedme, 2022-08-24:
Thanks for the reply. :)

I forgot to mention my system information.

OS: Manjaro Linux (Arch Linux)
Python Version: 3.10.5

Installing from pip would be great and quite easy.
If it's possible, please consider adding the support for the latest Python
Version.

Meanwhile, I will try installing using the suggestions in #25.

jgerstmayr, 2022-08-24:
I uploaded Python 3.9  and 3.10 manylinux wheels on pypi. It works on Ubuntu, but should also work on other linux distributions. Try again with 
pip install exudyn --pre

dummifiedme, 2022-08-24:
Thanks, I'll check them out.

closed 2022-09-01 by dummifiedme