jgerstmayr/EXUDYN #25: Unable to install the package

dummifiedme, 2022-08-24:
After #24 , I tried to install using the setup.py, but it threw an error relating to permissions. (Should I run it with sudo?).
Then I tried to change install directory using `install-base` flag, but it failed quoting "install is deprecated, use build or pip instead". 

I then tried to download a linux wheel and manually copied the unpacked files to python site packages. While trying to import exudyn, it threw error "no module named exudynCPP".

Can you please suggest a way to install the package, I really wish to start working with it! :)

jgerstmayr, 2022-08-24:
This very much depends on your Python installation. On ubuntu 18 and 20, default Python requires to run as "sudo python3 setup.py install".
BUT you can switch to a installation for user only: "python3 setup.py install --user"

I recommend to use virtual environments in Python. If they are in your user directory, also setup.py will not need su rights. You can also install different Python versions for the environments, allowing you to be compatible with Exudyn wheels (e.g. Python 3.8)

Copying the wheel for a wrong Python version can not work. At least, there must be the same Python version and 32/64 bits.

Please let me know, if you could resolve this problem. Please also follow the detailed explanations in theDoc.pdf !

dummifiedme, 2022-09-01:
Thanks for the all the help.

closed 2022-09-01 by dummifiedme