jgerstmayr/EXUDYN #28: SolutionViewer

MichaelUIBK, 2022-12-19:
When loading solution from multiple computed static problems the SolutionViewer does not correctly load the solution.
There are several steps shown where there is no displacement.

jgerstmayr, 2022-12-21:
LoadSolution() does currently not work with comments in between multiple steps. 
For workaround set:
    simulationSettings.solutionSettings.writeFileFooter = False
and after first SolveStatic() or SolveDynamic(), set:
    simulationSettings.solutionSettings.writeFileHeader = False
This will create consistent solution files.

closed 2022-12-21 by jgerstmayr

jgerstmayr, 2022-12-22:
with version 1.4.60.dev1 this issue is fully resolved; the SolutionViewer now accepts any comment lines in solution; writing of initial values can be turned off consistently by according solutionSettings.writeInitialValues flag.