jgerstmayr/EXUDYN #33: Debug

jgerstmayr, 2023-01-31:
However, another issue occurs! The main *.py file cannot be debugged through "main_sln_Template.sln". The error "The pdb file cannont be found" shows in the VS2017. Could you help me solve the new issue?

_Originally posted by @Jiang-Cui in https://github.com/jgerstmayr/EXUDYN/issues/31#issuecomment-1410596876_

jgerstmayr, 2023-01-31:
Make sure that:
- you have all .pdb files in the according Debug directories (in my case this is main/bin/x64/Debug)
- vc141.pdb (in VS2017)
- exudynCPP.pdb in the Debug directory
- Maybe you need to adjust settings, as recommended by MSVC for debugging Python-C++ codes (native debugging)
- check that you have installed all relevant MSCV components for debugging

Jiang-Cui, 2023-02-01:
Thank you, Professor! I debug sucessfully thourgh the VS2022, although little problem with no impact on the debug. I will keep your suggestions in my mind, and I will try them later.

closed 2023-02-02 by jgerstmayr