jgerstmayr/EXUDYN #34: Debug of Exudyn

Jiang-Cui, 2023-02-08:
Dear professor,

I re-installed the Exudyn today, but an error occured. Whe I click the "debug" in VS2022, the python interface just flashes by. When I click "start execution without debugging", the following error are shown. 

[image]

I use VS2022 and python 3.9 to debug. I have successfully installed Exudyn before, and the same procedure are repeated today but I can debug.

jgerstmayr, 2023-02-08:
I unfortunately have no access to VS2022, I only use VS2017 because this is the only version that is reliable regarding Eigen and other numerical libraries.

If you did not make any changes to Exudyn, then it is a migration problem to VS2022.
You will have to adapt the project settings if you migrate from VS2017 to VS2022. It may also be necessary to use another way to load exudynCPP.pyd . The current way it is done is based on the VS2017 recommendations.

It may also be the case that there is a compilation error. If this happens (e.g. if not all functions are correctly compiled in the .pyd DLL, it will show the exact error that you show). More information can be usually obtained under linux (e.g. using WSL).

closed 2023-02-08 by jgerstmayr

Jiang-Cui, 2023-02-09:
Thank you, Prof. Johannes Gerstmayr!