jgerstmayr/EXUDYN #35: The debug of NGsolveCraigBampton.py using VS2017

Jiang-Cui, 2023-02-15:
Dear Prof. Johannes Gerstmayr,

I have succesessfully built and installed Exudyn through VS2017 and py3.7. The debug of the common py files have realized. But there exists a issue for the debug of py files containing ngsolve such as NGsolveCraigBampton.py. NGsolveCraigBampton.py can run through py3.10 and py3.9, but the py3.10 and py3.9 do not support VS2017. I also try "pip install ngsolve --pre" in py3.7, and it cannot work. How could I debug NGsolveCraigBampton.py in VS2017?

Thank you very much!

Jiang-Cui, 2023-02-16:
Professor, I know how to debug NGsolveCraigBampton.py. Just use py3.9 or py3.10 to generate the meshdata, and then use py3.7 to directly process the data and we can debug NGsolveCraigBampton.py in VS2017. Thank you!

jgerstmayr, 2023-02-16:
got you! that makes sense as ngsolve only works with newer Python and compilation only runs under VS2017. But why not debug directly under Python?

Jiang-Cui, 2023-02-16:
I want to see how the code of CraigBampton method works. Thank you, Professor!

closed 2023-02-18 by jgerstmayr