jgerstmayr/EXUDYN #38: Getting started video with pip install?

jgerstmayr, 2023-03-24:
### Discussed in https://github.com/jgerstmayr/EXUDYN/discussions/37

<div type='discussions-op-text'>

<sup>Originally posted by **BradMBD** March 24, 2023</sup>
Dear Prof. Gerstmayr, I found EXUDYN now cannot be installed in the way you showed in the getting started youtube video (Aug. 2020). I installed the Anaconda3-5.2.0-Windows-x86_64 (Python 3.6.5) as mentioned in theDoc.pdf. Then, 

I first tried PIP INSTALLER. It turns out it doesn't work. I tried to import exudyn, the spyder console shows "ModuleNotFoundError: No module named 'exudyn'".

Then, I downloaded the "exudyn-1.6.0-cp36-cp36m-win_amd64.whl" that matches the python version. I installed via wheel files according to the tutorial in theDoc.pdf. I tried to import exudyn again in Spyder, it shows 

"ImportError: Import of exudyn C++ module failed; check 32/64 bits versions, restart your iPython console or try to uninstall and install exudyn". But, I am sure my labtop is 64bit and the version of Anaconda/python and exudyn are consistent (the recommended one in theDoc.pdf). So, could you please help give any suggestions?

Or, could you please make a updated getting started video tutorial?

Thanks,

Tengfei</div>

jgerstmayr, 2023-03-24:
you are right, the video for installation is outdated. There should be a new one.

Regarding the import error, I need to know more about your platform (Windows, etc.) and CPU. Reasons are given in FAQ and troubleshooting of theDoc.

In the current version, I recommend (and use) Anaconda 2021-11, 64bit, Python 3.9

BradMBD, 2023-03-26:
Dear Prof. Gerstmayr,

Thank you so much for your help. Using Anaconda 2021-11, 64bit, Python 3.9 you suggested. And I install the correponding whl file. Now I successfully run 'myFirstExample.py'. 

Btw, if I want to use the C++ source code, VS2017 is still now recommended as mentioned in theDoc.pdf?

Thanks,

Tengfei

jgerstmayr, 2023-03-28:
VS2017 is still the code I use. There may be alternative ways, however, there were problems with Eigen / the sparse solver  in VS2019: giving wrong results. 
I have not tested VS2022 so far.
People have also used linux successfully.

closed 2023-03-28 by jgerstmayr