jgerstmayr/EXUDYN #44: On the usage of a list as a default parameter

ManuelZ, 2023-04-26:
I've noticed that in several functions, a list is being used as a default parameter.

jgerstmayr, 2023-04-27:
Thanks for the hint. I raised an issue for that. There are around 30 such list initialization, which need to be checked if the args are modified internally and checked whether it makes sense to change them to None. This would then mean that users do not intuitively see how to initialize these args.

ManuelZ, 2023-04-27:
The arguments being modified internally isn't the only occasion where problems can arise. A generic example, this time with this happening inside a method:

```
class A:
    def __init__(self, x=[0, 0, 0]):
        self.x = x

a1 = A()
print(f"a1.x started with: {a1.x}")

a1.x.append(2)
print(f"a1.x was modified to : {a1.x}")

a2 = A()
print(f"a2.x started with: {a2.x}")
```

Prints:
```
a1.x started with: [0, 0, 0]
a1.x was modified to : [0, 0, 0, 2]
a2.x started with: [0, 0, 0, 2]
```

A concrete example of something like that happening in Exudyn would be if somebody tries to modify attributes by hand:
```
from exudyn.itemInterface import NodePoint

node1 = NodePoint()
print(node1.referenceCoordinates)

node1.referenceCoordinates[2] = 5
print(node1.referenceCoordinates)

node2 = NodePoint()
print(node2.referenceCoordinates)

```
Prints:
```
node1 started with reference coordinates:  [0.0, 0.0, 0.0]
node1 reference coordinates were modified to:  [0.0, 0.0, 5]
node2 started with reference coordinates:  [0.0, 0.0, 5]
```

A solution could look like:
```
class A:
    def __init__(self, x=None):
        self.x = x or [0, 0, 0]

a1 = A()
print(f"a1.x started with: {a1.x}")

a1.x.append(2)
print(f"a1.x was modified to : {a1.x}")

a2 = A()
print(f"a2.x started with: {a2.x}")

a3 = A([0, 0, 0])
print(f"a3.x started with: {a3.x}")

a1.x.append(9)
a2.x.append(9)
print(f"After modifying a1 and a2, a3.x is: {a3.x}")
```

Which prints:

```
a1.x started with: [0, 0, 0]
a1.x was modified to : [0, 0, 0, 2]
a2.x started with: [0, 0, 0]
a3.x started with: [0, 0, 0]
After modifying a1.x and a2.x, a3.x is: [0, 0, 0]
```

jgerstmayr, 2023-04-28:
These cases just reflect the basic problems of Python: the undefined behavior of special classes regarding copy and referencing. We need the default parameters in the itemInterface, as it alleviates the transfer of data to the items with the autocompletion in VSCode or Spyder. Otherwise, the itemInterface would not be needed and the pure dictionaries of e.g. AddObject(...) could be used as well.

I see an alternative by copying the lists in the constructor. This is roughly 3x slower. As they are flat, a pure copy.copy(...) should be sufficient. This could be done automatically when creating itemInterface.py.

ManuelZ, 2023-04-30:
The Python Language Reference should have all the needed information, although in a very dry state. The most clear explanations regarding copy and referencing that I've found are in places like [this](https://nedbatchelder.com/text/names.html), particularly around the point "Fact: Python passes function arguments by assigning to them" and [this Stackoverflow answer](https://stackoverflow.com/a/986145/1253729).

Another option would be to start using type annotations, either embedded in the code:

```
class NodeRigidBody2D:
    def __init__(self,
        name: str = '',
        referenceCoordinates: list = None,
        initialCoordinates: list = None,
        initialVelocities: list = None,
        visualization: dict = None):
    ...
```

 or in .pyi files, as [Numpy](https://github.com/numpy/numpy/blob/main/numpy/fft/_pocketfft.pyi) does. 

Scikit-Learn [doesn't yet support this](https://github.com/scikit-learn/scikit-learn/issues/16705), as it creates complications that in the short term may not be worth it, due to the lack of man-power (but that may be worth it in the long-term, specially when a type-checker leverages them).  The IDE suggestion wouldn't be a `[0, 0, 0]` list, for example, but plainly, a list.

jgerstmayr, 2023-05-02:
Let's put it simpler: how can you know, if values are copied or referenced in exudyn? Well, it is only defined by the CPP code, but you cannot see this from the Python function. 
Therefore, modifying any mutable arg inside a function is always a problem, no matter if this is a default arg or not.
As mentioned, I created issues (1536,1540) to test copying list arguments to surpass this problem mainly in the itemInterface. In performance-critical cases, this may be problematic, but otherwise this will be done.
NOTE: in your example, referenceCoordinates would become None and I am not attempting to implicitly convert None with pybind11 to a list in the cpp interface. Furthermore, it totally hides the default values. Default values are very helpful, also ChatGPT likes them a lot in order to better understand writing Exudyn code!

jgerstmayr, 2023-06-21:
Starting with version 1.6.173 (soon on github), all adaptations for mutable default arguments have been finished. Some structures have been replaced by None; However, using the automatic interface setup, it is ensured that lists, vectors, matrices, etc. are always copied in itemInterface; performance is not affected here, even if we initialize 100.000 items; in Python functions, lists are never modified inside a function, except that it is necessary and explicitly mentioned (but very rare).
In general, this should make pitfalls of default arguments nearly impossible.

The output of the code (which is not the recommended way to be done in Exudyn!):
```
import exudyn
print(exudyn.GetVersionString())
from exudyn.itemInterface import NodePoint

node1 = NodePoint()
print(node1.referenceCoordinates)

node1.referenceCoordinates[2] = 5
print(node1.referenceCoordinates)

node2 = NodePoint()
print(node2.referenceCoordinates)
```
now becomes the correct one:
```
1.6.171.dev1
[0. 0. 0.]
[0. 0. 5.]
[0. 0. 0.]
```
This problem should be resolved for now, but I am grateful about further feedback!