jgerstmayr/EXUDYN #47: VSCode and WSL(Ubuntu20.04LTS) visualization crash with mouseklick (left and right)

slyandsmart, 2023-05-31:
WSL with Ubuntu20.04LTS on Windows10 
VSCode: 1.78.2
Python: 3.8.10
Exudyn:  1.6.119.dev1

Left and right mouse klick on the visualization window crashes the simulation. 
No error code given. 

Setting following disables the mouse klick for selection and leads to no crash afterwards. 
    SC.visualizationSettings.interactive.selectionLeftMouse=False
    SC.visualizationSettings.interactive.selectionRightMouse=False

Zoom and view in the simulation is still possible.

jgerstmayr, 2023-06-01:
It seems to be related to WSL version and possible the viewer mode. Can you add WSL version (possibly try updating WSL)? Which viewer do you use (standard WSL?)
stable behavior is observed if you use (add this to .bashrc):
export DISPLAY=:0
export LIBGL_ALWAYS_INDIRECT=0
The latter command slows down visualization for larger models or shadows but is rather stable. 

Simply try visualization with:
import exudyn
exudyn.demos.Demo2()

jgerstmayr, 2023-06-08:
for now, it seems to be the best solution to either use Windows or pure linux or to set
SC.visualizationSettings.interactive.selectionLeftMouse=False
SC.visualizationSettings.interactive.selectionRightMouse=False

jgerstmayr, 2023-06-19:
This seems to be caused by OpenGL; resolved by setting:
export LIBGL_ALWAYS_SOFTWARE=0
preferably, put this into your .bashrc

closed 2023-06-19 by jgerstmayr