jgerstmayr/EXUDYN #49: Dear author, I encountered a question when simulate the example in this project. May I have your help

jianyshin, 2023-07-01:
Dear author, I encountered a question when simulate the example in this project. In detail, when I used the "exudyn" in pycharm, simulation of the test example "[flexibleRotor3Dtest](https://github.com/jgerstmayr/EXUDYN/blob/master/main/pythonDev/Examples/flexibleRotor3Dtest.py)" will cause the error "AttributeError: 'exudyn.exudynCPP.MainSystem' object has no attribute 'SolveDynamic'" in pycharm. I have no idea with this problem and will be approciate for you if some suggestions  heared from you. 
Thank you again!

jgerstmayr, 2023-07-04:
you should always provide exudyn and Python version (exudyn.GetVersionString(True) ). If you use mbs.SolveDynamic(...) you need to update to the newest exudyn version 1.6.119 or 1.6.164, using 
`pip install exudyn --pre -U`
If you updated already to your latest exudyn version, it may be a PyCharm limitation and you should change mbs.SolveDynamic(...) to SolveDynamic(mbs, ...).

jianyshin, 2023-07-05:
Thank you for your attention. This issue has been addressed completely.

closed 2023-07-05 by jianyshin