jgerstmayr/EXUDYN #50: ModuleNotFoundError

Jiang-Cui, 2023-07-05:
Dear professor,

Before, I have successfully installed Exudyn and use the C++ source code in the platform of Windows 10, VS2017, and python 3.7.9.

I now install the Exudyn in Windows 11 using the same steps using VS2017 and python 3.7.9. The cpp project can be successfully  compiled, but an error occured in debugging process. When I click "start execution without debugging" to debug the pytest.py, the following error are shown. This same error occurs before when I used VS2022. 
[image]

Coud you help me to solve this error? Thank you very much!

Jiang-Cui, 2023-07-05:
Thank you very much, Professor! I have copied the new exudynCPP.pyd to the file where the pytest.py is located, but the error occurred in Windows 11. Is there another way to solve the error?

Jiang-Cui, 2023-07-05:
Dear Professor,

I have simplify the __init__.py, and add print(sys.version). It seems that this cannot work, as shows

[image]

jgerstmayr, 2023-07-05:
Now you can just check if 3.7 64bits agrees with your compiled .pyd version (check your VS2017 settings). 

Another possible issue is that you changed something in exudyn. Did you? Commonly, if you are doing something wrong in the interface (e.g. failing to supply a API-function in the library correctly), this will raise a DLL load failed message without further information.   A workaround is to use a linux compilation, where gdb shows the missing API function.

Otherwise, it is a problem with directory settings in VS2017/Windows11 and you have to look there to resolve problems. This is not an exudyn problem anymore.

closed 2023-07-11 by jgerstmayr

Jiang-Cui, 2023-07-12:
Dear Prof. Gerstmayr,

Thanks for you reply. I have downloaded the latest code from github, the "ModuleNotFoundError" problem dose not exist. Thank you very much!