jgerstmayr/EXUDYN #62: Documentation mentions offsetUserFunction as if it was part of ObjectConnectorCoordinateVector

ManuelZ, 2024-04-13:
The documentation of `ObjectConnectorCoordinateVector` refers to `offsetUserFunction_t` and `offsetUserFunction` (in the description of `velocityLevel`), but those are not part of  `ObjectConnectorCoordinateVector`. Such parameters are only part of `ObjectConnectorCoordinate`. Further down in point 8.7.3.4 there are more references to such parameters, so I don't know if I'm misunderstanding something.

jgerstmayr, 2024-04-14:
You are right. This was the functionality of an old version and remained in the docu. Will be fixed soon.
Thanks!

closed 2024-04-21 by jgerstmayr

jgerstmayr, 2024-04-21:
There is still an open issue with the `ObjectConnectorCoordinateVector`, see list of issues #1032. At least the documentation has been cleaned up regarding the docu in version 1.8.21.