jgerstmayr/EXUDYN #77: ModuleNotFoundError: No module named 'exudyn.graphics'

damengziuu, 2024-08-11:
when I using exudyn.graphics , it shows moduleNotFoundError.But I only use import exudyn as exu, it is worked.
my code:
```
import exudyn as exu
import exudyn.graphics as graphics 
```

Error information:
```
[Running] python -u "e:\Git-Clone\EXUDYN-master\main\pythonDev\Examples\test.py"
Traceback (most recent call last):
  File "e:\Git-Clone\EXUDYN-master\main\pythonDev\Examples\test.py", line 2, in <module>
    import exudyn.graphics as graphics #only import if it does not conflict
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'exudyn.graphics'

[Done] exited with code=1 in 0.618 seconds
```

damengziuu, 2024-08-13:
I downloaded version 1.8.0 and it ran successfully

closed 2024-08-13 by damengziuu

jgerstmayr, 2024-08-14:
Note that the newest version 1.8.52 already uses the new module `exudyn.graphics` (but you have to install this with `pip install exudyn --pre`. However, the latest release 1.8.0 does not have this module. So you have to use the "older" examples, e.g., going back to 1.8.0 on github or readthedocs and take the description and examples from there.
Whenever 1.9.0 is released, everything should work smoothly with regular pip installers.

damengziuu, 2024-08-14:
Thanks for your reply, I am learning the sample corresponding to version 1.8.0。