jgerstmayr/EXUDYN #79: how  to setup joints speed not torque?

lizhuang2511, 2024-09-26:


lizhuang2511, 2024-09-27:
i can use objectconnectortorsionlspringdamper to setup joint velocity，but i can't understand rotationmark0,rotationmark1,can you explain to me?, in addition  code dependency where can see。

jgerstmayr, 2024-09-27:
You should read the documentation.
Use search method on readthedocs to find all examples using TorsionalSpringDamper ...
The TorsionalSpringDamper acts around the local Z-axis. If you want the TorsionalSpringDamper to act around the local X-axis, you have to rotate the local frames of your markers using rotationMarker0 (or 1) such that the X-axis becomes the Z-axis. This is the same for all joints using rotationMarker parameters (like GenericJoint, RevoluteJointZ, etc.).

closed 2024-10-06 by jgerstmayr