jgerstmayr/EXUDYN #88: Custom compilation

peter-manzl, 2024-12-20:
I tried compiling the current source code of Exudyn 1.9.0 by myself using
python setup.py install --parallel 
as well as
pip wheel . -v -w dist --no-deps
Both of these options finish compiling without an error message and when I check the version (pip show exudyn) shows the correct version. But when running various scripts from the examples
mbs.SolveDynamic(simulationSettings = simulationSettings)
fails. I tried different scripts, among others the [rigid body tutorial](https://github.com/jgerstmayr/EXUDYN/blob/master/main/pythonDev/Examples/rigidBodyTutorial.py). It does not matter if the graphics are started before calling solvingDynamics. When calling the script from the console no error message is shown, while Syder 5.4 shows "Windows fatal exception: access violation." before restarting the kernel. 

* Exudyn Version 1.9.0
* Python 3.9 and Python 3.11
* reproducible on a second machine
* Windows 10, compile with AVX2, 64 bits, pybind11-2.13.6
*  Compiler: msvc (Microsoft visual studio C++), version 14.40.33807

jgerstmayr, 2024-12-20:
The local builds (with successful run in Spyder) are done with MSVC 14.37.32822 (MSVC 2022 version). 

Please also provide NumPy version. I am using NumPy 1.26.4 and pybind11-2.13.6

It could also be an issue of Spyder (connected to the MSVC compiler). I am using Spyder 5.5.1. 
I remember similar problems in the past when using internally compiled .pyc files from MSVC. Due to missing debug information between Python and C++, this is hard to track. I also remember some Python packages that do not run inside Spyder without knowing the exact reason.

peter-manzl, 2024-12-20:
I tested both numpy versions 1.26.4 and a newer one - this did not change anything with this issue. 
The problem happens both when calling the script using spyder (both version 5.4.3 and 6.0.1) as well as from the command line (anaconda prompt). It changes how the error is printed/outputted but it occurs in all configurations. 

I will also try the other compiler.

SebiWeyrer, 2024-12-20:
I had the same problem. I have now updated my Visual Studio 2022 installation to 17.12.3 (which includes MSVC 19.42.34435). Now, when I compile the source code of Exudyn by myself on my local machine with `python setup.py install --parallel`, everything works perfectly. It seems like updating the VS 2022 installation fixed the problem for me.

- EXUDYN 1.9.0
- NumPy 1.26.4
- Python 3.11.5
- MSVC 19.42.34435 with VS 2022 17.12.3
- Windows 11 (24H2), 64 bits

jgerstmayr, 2024-12-28:
I checked the compilation with github actions. Unfortunately it only records the major and minor version number, which was 14.38 for exudyn version 1.8.0.
In version 1.9.0 github actions used MSVC 14.41 (the newer Microsoft Visual Studio 2022 toolset) which also runs fine on my side in Spyder and in the test suite. 
Check also a new Python conda environment, as there may be interference with other packages. In all cases, Numpy 1.26.4 was used, this should not be the issue. Certainly, it may also be a Spyder issue.

Please post configurations which worked.

peter-manzl, 2025-01-09:
I have now found the time to update the MSVC compiler to the newest version. This solved the issue and I did not encounter the error anymore, neither in Python version 3.9 nor 3.11.

closed 2025-01-09 by peter-manzl