jgerstmayr/EXUDYN #89: Computation of the derivative of the velocity constraints w.r.t. ODE2 coordinates

stehol01, 2024-12-26:
Im implementing a user integrator using SetUserFunctionNewton(). Within the user integrator, I would like to build the Jacobian by myself. For assembling the Jacobian, the derivative of the velocity constraints w.r.t. the ODE2 coordinates Zt is required. I tried to use the following code for that:

mainSolver.ComputeJacobianAE(mbs, scalarFactor_ODE2 =1., scalarFactor_ODE2_t=0., scalarFactor_ODE1=0, velocityLevel=True)
Zt = mainSolver.GetSystemJacobian()[nODE2:,0:nODE2]

Using Exudyn version 1.9.0, Im encounter the following User ERROR:

CSystem::JacobianAE_ODE2: velocityLevel=true not implemented

Is there a way to resolve this somehow in the current version?

Thanks for the help.

jgerstmayr, 2024-12-28:
Only the jacobian for position level constraints is implemented for joints and constraints. The index 2 solver uses the same position level constraint jacobian for holonomic constraints, as it is the same as the jacobian for velocity level constraints w.r.t. to velocity coordinates. For non-holonomic velocity level constraints, the velocity level jacobian is directly implemented and is not affected by the velocityLevel flag.

This means, there is no access to derivatives of velocity level constraints w.r.t. position coordinates. There is the option for newton.numericalDifferentiation.forAE = True, which works for the dense solver and which should do everything correctly.

stehol01, 2025-01-07:
Great. Thanks for the clarification.

closed 2025-04-01 by jgerstmayr